gaoxiang12/slam_in_autonomous_driving #4: My steps to go through compiling the repo

yuconglin, 2023-02-12:
I installed the following libraries before running cmake and make to build the whole repo in Ubuntu 20.04:

ROS Noetic (http://wiki.ros.org/noetic/Installation/Ubuntu)
pcl-ros (if you didn't choose to install ros-noetic-desktop-full): sudo apt install ros-noetic-pcl-ros
velodyne-msgs: sudo apt install ros-noetic-velodyne-msgs
opencv: sudo apt install libopencv-dev
glog: sudo apt install libgoogle-glog-dev
eigen3: sudo apt install libeigen3-dev
suitesparse: sudo apt install libsuitesparse-dev
pcl: sudo apt install libpcl-dev
yaml-cpp: sudo apt install libyaml-cpp-dev
tbb: sudo apt install libbtbb-dev
gmock: sudo apt install libgmock-dev
the system libraries above all together: sudo apt install -y libopencv-dev libgoogle-glog-dev libeigen3-dev libsuitesparse-dev libpcl-dev libyaml-cpp-dev libbtbb-dev libgmock-dev
Pangolin: https://github.com/stevenlovegrove/Pangolin
g2o: use the one in the thirdparty folder

I am hoping this summary can help others. BTW, I can testified that the building won't go through in Ubuntu 18.04 (as Xiang Gao suggested). I had to install Ubuntu 20.04 to build the repo.

AFEICHINA, 2023-02-12:
> I can testify that the building won't go through in Ubuntu 18.04

可以参考下faster-lio中的处理方式。1升级gcc9.0，2添加tbb2018_20170726oss到工程中，同时修改cmake/packages.cmake文件。

gaoxiang12, 2023-02-12:
The main problem is the <execution> in C++17. As I know you need gcc-8 or higher, and in Ubuntu 18.04 the default gcc is 7.5. The simplest way is to install a higher version of gcc and link all the outputs to a thirdparty tbb library.

yuconglin, 2023-02-12:
I was able to deal with gcc version, but didn't know how to deal with tbb issue. Here is the link from faster-lio issues' comment: https://blog.csdn.net/weixin_47552638/article/details/123724463. Thanks for sharing @AFEICHINA.

closed 2023-02-12 by yuconglin

teachermaDSM, 2023-09-05:
> > I can testify that the building won't go through in Ubuntu 18.04
> 
> 可以参考下faster-lio中的处理方式。1升级gcc9.0，2添加tbb2018_20170726oss到工程中，同时修改cmake/packages.cmake文件。
Can you be more specific？ i have update gcc to version 9.4, how to modify cmake/packages.cmke files