gaoxiang12/slam_in_autonomous_driving #11: 第二章的一些意见: 第一部分

yuconglin, 2023-02-13:
Page 15: (e1, e2, e3) 需要注明下 e1, e2, e3都属于R_{3} (下标3 ) 吗？
Page 17: 图2-1要注明来源吗? https://www.cvlibs.net/publications/Geiger2013IJRR.pdf
Page 17: 式 2.10 需要注明下 T_{wb} 的具体形式吗？比如建议读者参考式(2.82).虽然可能大部分读者都知道。
Page 17: 式 2.9 如果加一个 Barfoot书中文翻译版图6.2那样的图的话，感觉可以帮助一些初学者读者更好地理解。
Page 18，22: 这里直接提到了李代数，可能对于没有看过slam 14书的读者比较费解。不知道是不是需要加个注解，说李代数在slam 14书的第四讲有具体描述。
Page 18: 这里提到了不同的四元数定义，不知道是否需要注解下文献【10】的第三节有详细描述。
Page 23: 提到了以上内容在14讲里展开介绍过，但可能会让没有读过14讲的读者不理解这里说的“以上内容“具体包括哪些，是2.1.3节呢，还是整个2.1节。
Page 26: 括号里面的 “优化函数的目标更新量“ 对于有些缺乏这方面经验的读者来说可能有些突兀。可能要不然多加些解释，要不然直接去掉。等后面讲优化有关的内容的时候再提到这节的内容。
Page 26: 式 2.72不知道能否加个注解，解释下这个等式是可以推倒出来的，推倒过程可以见文献 [10]的(198)式。个人感觉这样会让读者理解容易些。同时个人感觉这里的½是由于式(2.69)的定义导致的。
Page 27: 不知道是否需要解释下式(2.78)的倒数第二个等号用到了一阶泰勒展开。
Page 30: 式(2.98) 我个人不太理解是如何推倒出来的，这里的w感觉和式(2.69)是一样的，所以不知道这里右雅克比是如何出现的。

谢谢高博。

FengyuGuo, 2023-02-13:
第二章开头就说了，很多部分参照14讲部分，所以我觉得现在的架构已经比较合理了。如果真的要全部说清楚，差不多要把整本barfoot的书塞进来。

gaoxiang12, 2023-02-14:
- 第2条加了引用
- (2.72)事实上就是(2.58)，只是把omega定义统一了。
- (2.78)后面现在加了一句“从泰勒展开”的说明。
- (2.98)确实是多打了一个J_r，已经修复。

closed 2023-02-14 by yuconglin