gaoxiang12/slam_in_autonomous_driving #12: compilation error

qlibp, 2023-02-13:
The compilation order of CMake's `add_subdirectory()` may result in compilation error.

[image]

`velodyne_msgs` are found by catkin, the cmake invoke compilation on tools folder before common folder, which results in the following error(I guess)

qlibp, 2023-02-13:
One tricky way to handle this without modifying the CMake code, it's to run `make -j4` multiple times until cmake finishing building the message...

gaoxiang12, 2023-02-13:
Looks like I should let the sad.common depend on lidar messages. I'll look into it.

qlibp, 2023-02-13:
Already verified, once I comment out the `add_subdirectory(tools)` line first, and compile the code, the messages will generate successfully.

qlibp, 2023-02-13:
Also, can't build objects with g2o successfully. I try two ways to build the entire project.

1. build at the {project_src}/build, and trigger an error
2. I run `cmake ..` & `make` inside `thirdparty/g2o/build` and I can't build the whole project still.

qlibp, 2023-02-13:
> Also, can't build objects with g2o successfully. I try two ways to build the entire project.
> 
> 1. build at the {project_src}/build, and trigger an error
> 2. I run `cmake ..` & `make` inside `thirdparty/g2o/build` and I can't build the whole project still.

I don't know which `g2o/config.h` am I missing, I grep the entire folder and find out that it's within `thirdparty/g2o/build/`
So, I add an additional line of  `include_directory()` in `cmake/packages.cmake` to make it compiled.

If I get it wrong, plz correct me.

qiuzhichang, 2023-03-04:
sudo apt install ros-noetic-velodyne-msgs

qiuzhichang, 2023-03-04:
```
sudo apt install ros-noetic-velodyne-msgs

```

aqiuxx, 2023-03-04:
```

sudo apt install ros-noetic-velodyne-msgs

```

ForLoveandPeace, 2023-04-11:
Hi, you need to do `sudo make install` in g2o/build, unless CMake can't find the path.

ForLoveandPeace, 2023-04-11:
Hi, you need to do `sudo make install` in g2o/build, because CMake can't find the path.