gaoxiang12/slam_in_autonomous_driving #15: Provide Dataset link for foreigners

Taeyoung96, 2023-02-14:
@gaoxiang12   

Hi, Thanks for your great work!  
I want to test many algorithms in this repository with your provided bagfile.  
Unfortunately, Baidu driver is very difficult for foreigners to use. :cry:    

I'm sorry but, can you provide the data set using something like Google drive?  
If possible, it would be of great help to me and other foreigners.  

Regardless of my opinion, thank you for sharing a good repository. :+1:

gaoxiang12, 2023-02-15:
Hi @Taeyoung96 

I'm also thinking about sending the data to google drive, but it is also difficult for me to send about 270G data to google drive from China. Maybe I should ask some friends for help with a hard copy on disk. 

Anyway, most of the public datasets (NLCT, UTBM, ULHK, etc.) can be found on their own website. You can directly download them and convert them to rosbags. The convert scripts can be found in the FastLIO repo: https://github.com/hku-mars/FAST_LIO

gaoxiang12, 2023-02-15:
@Taeyoung96 Could you please try the onedrive link: https://1drv.ms/u/s!AgNFVSzSYXMahcEZejoUwCaHRcactQ?e=YsOYy2 
I'm not sure if it is working in Korea.

Taeyoung96, 2023-02-15:
@gaoxiang12   
Thanks for your pay attention to my opinion!  

I can download `outdoor_Mainbuilding` in `/avia` folder. It works in Korea!  :+1:  
However, there is no file on `/NCLT` and `/UTBM`.  
If you fill in the empty folders with data, I think the rest will be able to use it well.  

Thank you very much for listening to my opinion.

gaoxiang12, 2023-02-15:
Cool, I'm still uploading the data. It may last for a few hours or maybe a few days. Please check the link later.

closed 2023-02-15 by Taeyoung96