gaoxiang12/slam_in_autonomous_driving #17: 如何从原始数据集生成rosbag?

yuconglin, 2023-02-15:
我一开始试着从百度网盘下载，结果巨慢。没钱买超级会员。我想问问作者和其他读者，方便告诉一下如何自己生成某些rosbag吗？

我目前一些观察和猜测：
1. NCLT数据是下载对应日期的某些原始数据，然后通过那个python script转换吗？ 对于某个确定的session, 是需要所有的 *.tar.gz文件吗？
2. UTBM数据大部分可以从 https://epan-utbm.github.io/utbm_robocar_dataset/ 直接下载。
3. Avia数据可以从 https://drive.google.com/drive/folders/1YL5MQVYgAM8oAWUm7e3OGXZBPKkanmY1 直接下载。
4. ULHK数据 如何从 https://github.com/weisongwen/UrbanLoco 转换？
5. WXB 数据好像式可选项目，看起来不是必须的。
6. 2dmapping数据不大，我就慢慢下好了。

谢谢各位。

gaoxiang12, 2023-02-15:
大部分数据集有特定的转换工具，用python写的，可以参考fastlio2中的转换方法。
或者您有什么方便下载的工具吗？国内目前能用的大型网盘可能也就百度云或者微软的onedrive了。

yuconglin, 2023-02-15:
谢谢高博，我自己琢磨琢磨怎么转换。google drive如果维持这么大的空间也不是免费的。

closed 2023-02-15 by yuconglin

gaoxiang12, 2023-02-15:
请尝试一下OneDrive：https://1drv.ms/u/s!AgNFVSzSYXMahcEZejoUwCaHRcactQ?e=YsOYy2 
我已经把Onedrive链接更新到readme中了。

yuconglin, 2023-02-15:
目前好像几个偏大的文件夹里面我这边看起来还是空的，也许还在上传。我过段时间再看看。

wangg12, 2023-02-15:
> 大部分数据集有特定的转换工具，用python写的，可以参考fastlio2中的转换方法。 或者您有什么方便下载的工具吗？国内目前能用的大型网盘可能也就百度云或者微软的onedrive了。

阿里云盘应该也行

yuconglin, 2023-02-15:
试着下载了一个大一些的NCLT rosbag, 完全没问题。比百度网盘好太多。高博为读者考虑周全，感谢感谢。