gaoxiang12/slam_in_autonomous_driving #21: 第三章一些读后感和问题

yuconglin, 2023-02-19:
纯属个人观点，不一定对。

Page 39: “我们不必关心IMU的内部测量原因” 这里感觉该用 “原理”？
Page 52: 3.3.2节，这里的RTK设备也适用于普通的GNSS信号，不知道是否要注释一下？
Page 64: 式(3.36)第四个等号，delta_theta^后面的那个a好像该加个波浪帽？

https://github.com/gaoxiang12/slam_in_autonomous_driving/blob/master/src/ch3/utm_convert.cc#L31 这一行 const double& 没有必要用const reference， 应该直接double就可以达到一样的效果。

https://github.com/gaoxiang12/slam_in_autonomous_driving/blob/master/src/ch3/process_gnss.cc#L17 和https://github.com/gaoxiang12/slam_in_autonomous_driving/blob/master/src/ch3/run_eskf_gins.cc#L17 这里三行的三个参数是根据本书提供的数据固定的吗，还是读者可以自己设置？

https://github.com/gaoxiang12/slam_in_autonomous_driving/blob/master/src/ch3/process_gnss.cc#L79 这里是否需要注释下这个原点是如何确定的？

3.5.5 节我跑的结果(在不用轮速测量的情况下)看起来有两个地方发散更明显，我在两个电脑上跑的结果类似。不知道这是否是个问题。
以下是截图。
[image]

谢谢高博。

gaoxiang12, 2023-02-21:
感谢，文本方面的内容已经在2.21版本修复。您看得非常仔细啊。
代码相关的内容会在后面的push中更新。

closed 2023-02-21 by yuconglin

yuconglin, 2023-02-21:
和当年看14讲和Barfoot一样，抱着学习的态度看的。这本书提供的很多细节填补了这类教材的空白。感谢高博利用业余时间写书。