gaoxiang12/slam_in_autonomous_driving #23: 第四章读后感和问题

yuconglin, 2023-02-21:
基于2/19更新的版本。可能有关问题在2/21版本已经修复。纯属个人观点，不一定对。

Page 88: 式(4.6c) g 后面的Delta_t_{ij}^2 好像累加不对，即使考虑到下一页Delta_t_{ij}的定义。
Page 91:式(4.15)第二个等号，加法的第一项感觉下标打错了？好像第二项泰勒展开后的下标也打错了。
Page 92:式(4.19)等号左边，括号内感觉少了个负号？
Page 93:式(4.21)等号右边的R感觉少了波浪帽子？
Page 93:式(4.25)第二个等号累加上限感觉是j-2?
Page 96:式(4.35)第二个等号以及以后的，右边分出来的那项的Jocabian下标感觉是j - 1?
Page 96:式(4.36)第四，五个等号Delta_v_ij应该需要加个波浪帽子？参照式(4.13). 
左边感觉需要加{ij}的下标？
Page 96:式(4.37)第二，三个等号的左边第一项delta_{p}_{ij}应该用大的Delta?
第三个等号的第三项b_{g,i}前面需要加个小delta?
Page 98:图 4-1两边下标都是i，会不会是一侧是i，另一侧应该是j?
Page 99:式(4.42)第一个等号右边感觉应该是 (R_i * Exp(phi_i))^T ?
Page 105: “VertexVelocity” 的 “oplusImpl” 和实际代码有点点不同，虽然实现的结果一样。
Page 106: “EdgeInertial”的代码应该是被搬到的了 src/ch4/g2o_types.cc(h)
Page 99: 式(4.44) J_{b,r} 和 J_{r,b} 同时存在，但应该指的同一个东西。

谢谢高博。

wangg12, 2023-03-01:
page 88补充一点: "k 和 j", "k 至 j" 感觉这里k应该是i

gaoxiang12, 2023-03-02:
感谢楼上的指出，我这周在出差，等回北京了再修改。

gaoxiang12, 2023-03-06:
楼上反映的问题已经确认，3.6号的更新中已经修复。

closed 2023-03-10 by yuconglin