gaoxiang12/slam_in_autonomous_driving #28: make error

grdiv, 2023-02-24:
Is there a function named "jr" or "jr_inv" in SO3?

```
slam_in_autonomous_driving/src/ch4/imu_preintegration.cc:58:25: error: ‘jr’ is not a member of ‘SO3’ {aka ‘Sophus::SO3<double>’}
   58 |     Mat3d rightJ = SO3::jr(omega);  // 右雅可比
```

```
error: ‘Sophus::SO3Base<Sophus::SO3<double> >::SO3Product<Sophus::SO3<double> >’ {aka ‘class Sophus::SO3<double>’} has no member named ‘jr_inv’
  209 |         _jacobianOplusXi.block<3, 3>(0, 0) = (_measurement.so3().inverse() * v->estimate().so3()).jr_inv();  // dR/dR
```

gaoxiang12, 2023-02-25:
@grdiv 
There's no jr() in the original Sophus. This is an extension of this book: 
https://github.com/gaoxiang12/slam_in_autonomous_driving/blob/master/thirdparty/sophus/sophus/so3.hpp#:~:text=SOPHUS_FUNC%20static%20Transformation%20jr,(R.log())%3B%20%7D

grdiv, 2023-02-25:
i see

closed 2023-02-25 by grdiv