gaoxiang12/slam_in_autonomous_driving #44: some typo

zerolover, 2023-03-12:
- 14：最后加上回环检测，形成**完成**的 SLAM 系统。
- 5.3第1段：**传统在** SLAM 应用中，我们关心的元素往往是基本的
  
- 6.1第1段：然而，大部分**轮机器人**的运动方式某个固定的平面当中
  
- 173：就会涉及到光线投射算法（ray casting）和栅格化算法（rasterization），加
  
- 8.1第2段：与雷达里程计进行**融合合**时
  
- 9.4第1段：点云地图的质量很大程度**来来**自充分的回环检测
  322 却是对场景静态三维结构的一个很好的描述

zerolover, 2023-03-12:
第2章pdf已发送，有个地方备注一下
[image]
这里可以参考下面pdf的倒数第二页，以及图中出现的 SE3的^ 最好提一下
https://natanaso.github.io/ece276a2022/ref/ECE276A_14_SO3_SE3.pdf

gaoxiang12, 2023-03-13:
1 楼已经确认修复。
2 楼：这个不是一样的吗。。。 角速度不变，李代数速度项是个R^T v
hat的定义已经补充。

zerolover, 2023-04-02:
- 本书不再对诸如四元数到旋转矩阵变换之类的过程进行详细**的**展开
- 去掉下标 a
[image]
- 鉴于是在公式2.48之后才提到省略的记法，所以在2.46时不建议缩写
[image]

zerolover, 2023-04-02:
[image]
如果旋转矩阵的 Exp(w) 近似为 I + hat(w) 的话，也是存在这个问题的。我觉得这里的描述稍微有些问题。

[image]
同理，和SO3保持一致的话，四元数就不应该用近似的方式。
当然用近似的方式也行吧，只是这里的 cal{Q} 好像之前也没有出现过。

zerolover, 2023-04-02:
[image]

[image]

gaoxiang12, 2023-04-03:
已经修复，楼上那个加粗实际已经是粗体了，只是字体比较小看不太出来。

closed 2023-04-04 by zerolover