gaoxiang12/slam_in_autonomous_driving #45: 第七章读后感

yuconglin, 2023-03-12:
根据3/7更新版本。纯属个人见解，不一定对。不可否认我可能读得比较粗略，文字方面没有发现啥不对。主要是对于代码的小问题。

1. https://github.com/gaoxiang12/slam_in_autonomous_driving/blob/master/src/ch7/loam-like/feature_extraction.cc#L29 这里131是怎么得到的？

2. https://github.com/gaoxiang12/slam_in_autonomous_driving/blob/master/src/ch7/loam-like/feature_extraction.cc#L61 这里(int)多余？

3. https://github.com/gaoxiang12/slam_in_autonomous_driving/blob/master/src/ch7/loam-like/feature_extraction.cc#L83 这里point_info_count在++完之后好像就没啥用了？

4. https://github.com/gaoxiang12/slam_in_autonomous_driving/blob/master/src/ch7/loam-like/loam_like_odom.h#L51 这里的param其实并不存在？

5. https://github.com/gaoxiang12/slam_in_autonomous_driving/blob/master/src/ch7/loosely_coupled_lio/cloud_convert.cc#L91 这里为啥用curvature代指时间？

6. https://github.com/gaoxiang12/slam_in_autonomous_driving/blob/master/src/ch7/loosely_coupled_lio/measure_sync.h#L24 这里多了一个“时”？

谢谢高博。

wangg12, 2023-03-12:
0310版本：

- page 294: 
  - 7.5.2小节第1段话，“第3节”应为“第3章”; 
  - 为保持行文一致性，式（7.28）和（7.29）中Ln英为Log

gaoxiang12, 2023-03-13:
感谢，文本方面已经修复。

closed 2023-03-23 by yuconglin