gaoxiang12/slam_in_autonomous_driving #47: 轨迹评测

gongjun136, 2023-03-12:
高博你好，我正打算用evo评测精度，将所有pose导出后，发现位姿的数值都很小，这是增量ndt的轨迹，看完代码也不知道为啥：
[image]

gaoxiang12, 2023-03-12:
UI里显示程序跑起来了吗？

gongjun136, 2023-03-12:
跑起来了，并且显示的还是对的，我主要添加了如下代码，感觉并没有太大问题，添加了 这些代码后运行起来很慢，可能是我写入频率太高了？：
[image]
[image]

gaoxiang12, 2023-03-13:
你应该在程序开始运行时打开一个文件，每次往里写入一行。而不是在每个循环中重新打开这个文件然后追加内容。
写入的程序看上去没什么问题，你可以换种实现方式再试试，或者用glog打印一下。

gongjun136, 2023-03-14:
> 你应该在程序开始运行时打开一个文件，每次往里写入一行。而不是在每个循环中重新打开这个文件然后追加内容。 写入的程序看上去没什么问题，你可以换种实现方式再试试，或者用glog打印一下。

好的，已经做了如下的修改，但是发现还是挺慢的，其次我将/novatel_data/inspvax的位姿信息导出为东北天坐标系下的轨迹作为真值，把评测的轨迹导出作为估计值，发现它们轨迹z轴接近90度偏差，之前在用组合导航做真值时好像也遇到了这个问题，有点不解这是什么导致的？
[image]
[image]

gaoxiang12, 2023-03-14:
我不太确定这个数据集中的RTK是什么坐标系，角度是东北天还是北东地，另外它是否有安装外参，这些得去查数据集对应的网站或者论文。
所以处理各数据集真值是一件相当枯燥而且容易踩坑的事情。。。

gongjun136, 2023-03-14:
我直接自己手动旋转 90度了，感觉可能是安装参数有个90度：
[image]

但是我又碰到了gflag的bug，参数无法设置，于是我把所有参数在程序开始时打印出来，结果确实没有设置成功。
[image]
[image]

gaoxiang12, 2023-03-20:
@gongjun136 
gflags的 bool 类型需要写成 --display_map=false 的形式，不能中间带空格。
我也不知道为啥。。可能gflags就得这么用