gaoxiang12/slam_in_autonomous_driving #51: 运行./bin/test_loosely_lio 已放弃 (核心已转储) ，可能和Pangolin有关

GuaguaguaLiu, 2023-03-18:
高博，您好！在运行第7章代码`./bin/test_loosely_lio`时， 会报 “已放弃 (核心已转储)“的错误，用gdb看的时候发现断在了pangolin_window_impl.cc 的 `pangolin::FinishFrame() ;` （ 是通过pangolin_window.cc的第23行 `impl_->render_thread_ = std::thread([this]() { impl_->Render(); });`进去的）。感觉和Pangolin显示有关，继续运行了第8章`./bin/test_lio_iekf `, 也出现了同样的问题。我安装的Pangolin版本是v0.8 [https://github.com/stevenlovegrove/Pangolin/releases/tag/v0.8](url) （尝试了其他版本v0.7, v0.6应该都不对，编译不过去）。
[image]
[image]

gaoxiang12, 2023-03-18:
pangolin从第3章开始就用了，前面几章也这样吗？

GuaguaguaLiu, 2023-03-18:
测试了第3章和第4章，在我这里同样出现了上述问题。
第5章和第6章没有这个问题，因为没用到Pangolin。

gaoxiang12, 2023-03-20:
pangolin有时和测试的电脑配置有关。我有一台22.04的yoga pro 14s，带pangolin的程序起来就会死机，不开ui就没事。我怀疑pangolin用的opengl在硬件上有一些限制，但大部分台式机是可以运行的。

GuaguaguaLiu, 2023-03-21:
可能确实是，换了一台笔记本就行了。

UAVlover, 2023-04-28:
@GuaguaguaLiu  朋友，你好呀，你的笔记本型号是啥呀。我想在issue里收集一下，出现这类问题的机型，这样或许能更好的帮助别的朋友。

luameows, 2023-08-05:
运行前执行

ywwbehappy, 2023-11-07:
> @gaoxiang12 @GuaguaguaLiu 运行前执行 export LIBGL_ALWAYS_SOFTWARE=1 参考链接 [pangolin 黑屏，系统卡住，但pangolin example可以正常运行解决办法](https://blog.csdn.net/ABC_ORANGE/article/details/130652414)

确实有用，非常感谢

WangPei-hitman, 2024-01-13:
> pangolin有时和测试的电脑配置有关。我有一台22.04的yoga pro 14s，带pangolin的程序起来就会死机，不开ui就没事。我怀疑pangolin用的opengl在硬件上有一些限制，但大部分台式机是可以运行的。

22.04安装不了ROS1, 高博有什么办法呢？目前在docker里运行，但是不能支持显卡和opengl。电脑太新了，20.04的内核不支持很多硬件。。。感觉很矛盾

qhtxj, 2024-09-12:
> @WangPei-hitman 笔记本装2204和ros2，然后通过bridge把本书的包都转成ros2的，还需要改下cmake结构和rosbagio模块。这样应该可行？

Ubuntu22可以源码安装ROS1 noetic。