gaoxiang12/slam_in_autonomous_driving #53: Comments

GuaguaguaLiu, 2023-03-19:
根据文稿316的版本：

P139, 图5-2 建议说明r=Range。
P149, “把每次结果与之间的结果进行比较” ， “之间的”是否应为 “之前的” ？
P206, “既可以用于 scan to scan 匹配，也可以用于 scan-to-map 的配准” ， 前后的scan to …建议统一(是否使用小短线) 。
P215，“那么它的占据概率主会逐渐上升至1”， 是否多了“主”？
P223, “我们每次求解是的相对于当前子地图的位姿”， 应为“求解的是”。
P223, “一些关键帧拷至新的子地图中”， 是否是“拷贝”？
P280, 图7-10, 建议说明图中“l， j，m”的表示。
P285, 图7-12, 建议同样指出角点和平面点的颜色（如图7-11）。
P328, “仅有平移信息而不含位姿信息”， 这句话少了句号。
P195，运行./dataset/sad/2dmapping/test_2d_lidar.bag， 在下载的数据集中没有名为test_2d_lidar.bag这个文件，应为 "floor*.bag"。

谢谢高博！

gaoxiang12, 2023-03-20:
感谢指出，已经全部修复。

GuaguaguaLiu, 2023-03-27:
根据320版本，再补充几个小typos:
P11, ”而 L4 则依赖离线地图[32?]“, 文献显示有点问题；
P359, ”Map-based precision vehicle localization in urban environments.,“ 参考文献36可能多了个句号；
P341, 9.5 地图的导出与标注， 感觉上这一章没有提到做地图标注相关。