gaoxiang12/slam_in_autonomous_driving #54: 第四章 ./bin/test_preintegration 段错误 (核心已转储)

lijh1024, 2023-03-23:
第四章代码test_preintegration 运行到IMUPreintegration::Integrate(const IMU &imu, double dt)的
// 更新噪声项
 cov_ = A * cov_ * A.transpose() + B * noise_gyro_acce_ * B.transpose();

出现如下错误：
Program received signal SIGSEGV, Segmentation fault.
Eigen::internal::gemm_pack_lhs<double, long, Eigen::internal::const_blas_data_mapper<double, long, 0>, 8, 4, 0, false, false>::operator() (this=<optimized out>, blockA=0x7fffffffc330, lhs=..., depth=9, rows=9, stride=<optimized out>, offset=0) at /usr/include/eigen3/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1757
1757	        pstore(blockA+count, cj.pconj(A)); count+=PacketSize;

gaoxiang12, 2023-03-24:
这个不太看的出来错在什么地方。有编译环境之类的信息吗？

lijh1024, 2023-03-24:
> 这个不太看的出来错在什么地方。有编译环境之类的信息吗？

感谢高博的回复！我的系统是ubuntu20.04 编译环境信息如下：
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Found gmock: gmock and gtests will be built
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- livox_ros_driver: 2 messages, 0 services
-- Eigen3 version: 3.3.7
-- Eigen found (include: /usr/include/eigen3, version: 3.3.7)
-- FLANN found (include: /usr/include, lib: /usr/lib/x86_64-linux-gnu/libflann_cpp.so)
-- PCL version: 1.12.1
-- OpenCV version: 4.2.0
-- Pangolin version: 0.8
-- yaml-cpp version: 0.6.2

cmake ..编译时会出现如下警告
Cannot generate a safe runtime search path for target test_preintegration
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libpcl_kdtree.so.1.12] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_search.so.1.12] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_features.so.1.12] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_sample_consensus.so.1.12] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_filters.so.1.12] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_ml.so.1.12] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_segmentation.so.1.12] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_surface.so.1.12] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_common.so.1.12] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_octree.so.1.12] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_io.so.1.12] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.
pcl 库可能有问题，但是程序崩溃指向Eigen库，我想可能是我的环境有问题，我打算重装一下系统再试试
目前代码跑了第二至第六章，只有第四章这里有问题，后面章节数据集还没下载好

gaoxiang12, 2023-03-24:
不着急，你也可以先跑跑后边的程序，看有没有类似的问题。主要是我这边没法复现。

lijh1024, 2023-03-25:
> 不着急，你也可以先跑跑后边的程序，看有没有类似的问题。主要是我这边没法复现。

恩恩 好的高博。
**第七章的代码，其他的test没有问题, test_loosely_lio有问题:**
void LooselyLIO::Align() {
    FullCloudPtr scan_undistort_trans(new FullPointCloudType);
    pcl::transformPointCloud(*scan_undistort_, *scan_undistort_trans, TIL_.matrix());
    ...
}
代码在 pcl::transformPointCloud这句终止，出现如下报错：
I0325 15:37:20.671840 22907 loosely_lio.cc:53] call meas, imu: 10, lidar pts: 2666
I0325 15:37:20.671860 22907 static_imu_init.cc:73] mean acce: -0.00223703 0.000396294 000.0978845
I0325 15:37:20.671895 22907 static_imu_init.cc:95] IMU 初始化成功，初始化时间= 9.99019, bg = -0.00302408 00.00172013 0.000526504, ba = 000.221898 -0.0393097 00-9.70947, gyro sq = 7.24173e-05 4.26878e-05 3.73627e-05, acce sq = 7.51306e-07 1.98639e-06 1.85595e-07, grav = 00.224135 -0.039706 0-9.80736, norm: 9.81
I0325 15:37:20.671919 22907 static_imu_init.cc:99] mean gyro: -0.00302408 00.00172013 0.000526504 acce: 000.221898 -0.0393097 00-9.70947
I0325 15:37:20.671931 22907 loosely_lio.cc:97] IMU初始化成功
Failed to find match for field 'time'.
I0325 15:37:20.673998 22907 loosely_lio.cc:53] call meas, imu: 10, lidar pts: 2817
I0325 15:37:20.674010 22907 eskf.hpp:225] skip this imu because dt_ = 1.54771e+09
浮点数例外 (核心已转储)

gdb调试后，也是指向Align函数出了问题：
Thread 1 "test_loosely_li" received signal SIGFPE, Arithmetic exception.
0x00007ffff7f7b80e in pcl::PointCloud<sad::FullPointType>::assign<__gnu_cxx::__normal_iterator<sad::FullPointType const*, std::vector<sad::FullPointType, Eigen::aligned_allocator<sad::FullPointType> > > > (new_width=0, last=..., first=..., this=0x55555563b300) at /usr/include/c++/9/bits/stl_vector.h:915
915           size() const _GLIBCXX_NOEXCEPT
(gdb) bt
   #0  0x00007ffff7f7b80e in pcl::PointCloud<sad::FullPointType>::assign<__gnu_cxx::__normal_iterator<sad::FullPointType const*, std::vector<sad::FullPointType, Eigen::aligned_allocator<sad::FullPointType> > > > (new_width=0, last=..., first=..., 
    this=0x55555563b300) at /usr/include/c++/9/bits/stl_vector.h:915
#1  pcl::transformPointCloud<sad::FullPointType, double> (cloud_in=..., cloud_out=..., transform=..., 
    copy_all_fields=copy_all_fields@entry=true) at /usr/local/include/pcl-1.12/pcl/common/impl/transforms.hpp:232
#2  0x00007ffff7f72669 in sad::LooselyLIO::Align (this=0x7fffffffbca0) at /usr/include/c++/9/bits/shared_ptr_base.h:1020
#3  0x00007ffff7f73241 in sad::LooselyLIO::ProcessMeasurements (this=0x7fffffffbca0, meas=...)

**第八章 ./bin/test_lio_iekf出现了与第七章同样的问题：**
I0325 15:47:42.321244 24018 static_imu_init.cc:73] mean acce: -0.00215149 00.00016898 000.0978879
I0325 15:47:42.321276 24018 static_imu_init.cc:95] IMU 初始化成功，初始化时间= 9.99018, bg = -0.00259592 00.00176906 0.000707638, ba = 000.213411 -0.0167615 00-9.70973, gyro sq = 5.96793e-05 4.42613e-05 3.58264e-05, acce sq = 9.71749e-07 1.85436e-06 2.14871e-07, grav = 000.215562 -0.0169305 00-9.80762, norm: 9.81
I0325 15:47:42.321321 24018 static_imu_init.cc:99] mean gyro: -0.00259592 00.00176906 0.000707638 acce: 000.213411 -0.0167615 00-9.70973
I0325 15:47:42.321332 24018 lio_iekf.cc:151] IMU初始化成功
Failed to find match for field 'time'.
I0325 15:47:42.556319 24018 lio_iekf.cc:42] call meas, imu: 10, lidar pts: 3429
I0325 15:47:42.556402 24018 iekf.hpp:192] skip this imu because dt_ = 1.54771e+09
浮点数例外 (核心已转储)

gdb调试显示 LioIEKF::Align()函数 pcl::transformPointCloud()处 出了问题
void LioIEKF::Align() {
    FullCloudPtr scan_undistort_trans(new FullPointCloudType);
    pcl::transformPointCloud(*scan_undistort_, *scan_undistort_trans, TIL_.matrix().cast<float>());
    scan_undistort_ = scan_undistort_trans;
    ...
}

Thread 1 "test_lio_iekf" received signal SIGFPE, Arithmetic exception.
0x00007ffff7f386d5 in pcl::PointCloud<sad::FullPointType>::assign<__gnu_cxx::__normal_iterator<sad::FullPointType const*, std::vector<sad::FullPointType, Eigen::aligned_allocator<sad::FullPointType> > > > (new_width=0, last=..., first=..., this=0x55555563a820) at /usr/include/c++/9/bits/stl_vector.h:915
915           size() const _GLIBCXX_NOEXCEPT
(gdb) bt
   #0  0x00007ffff7f386d5 in pcl::PointCloud<sad::FullPointType>::assign<__gnu_cxx::__normal_iterator<sad::FullPointType const*, std::vector<sad::FullPointType, Eigen::aligned_allocator<sad::FullPointType> > > > (new_width=0, last=..., first=..., 
    this=0x55555563a820) at /usr/include/c++/9/bits/stl_vector.h:915
#1  pcl::transformPointCloud<sad::FullPointType, float> (cloud_in=..., cloud_out=..., transform=..., 
    copy_all_fields=copy_all_fields@entry=true) at /usr/local/include/pcl-1.12/pcl/common/impl/transforms.hpp:232
#2  0x00007ffff7f29221 in pcl::transformPointCloud<sad::FullPointType> (copy_all_fields=true, transform=..., cloud_out=..., 
    cloud_in=...) at /usr/local/include/pcl-1.12/pcl/common/transforms.h:237
#3  sad::LioIEKF::Align (this=0x7fffffffbc40)
    at /home/user/slam/slam_in_autonomous_driving-master/src/ch8/lio-iekf/lio_iekf.cc:79
 

**第八章 ./bin/test_lio_preinteg出现了和第四章同样的问题：**
I0325 15:56:19.972568 25538 static_imu_init.cc:73] mean acce: -0.00215149 00.00016898 000.0978879
I0325 15:56:19.972599 25538 static_imu_init.cc:95] IMU 初始化成功，初始化时间= 9.99018, bg = -0.00259592 00.00176906 0.000707638, ba = 000.213411 -0.0167615 00-9.70973, gyro sq = 5.96793e-05 4.42613e-05 3.58264e-05, acce sq = 9.71749e-07 1.85436e-06 2.14871e-07, grav = 000.215562 -0.0169305 00-9.80762, norm: 9.81
I0325 15:56:19.972618 25538 static_imu_init.cc:99] mean gyro: -0.00259592 00.00176906 0.000707638 acce: 000.213411 -0.0167615 00-9.70973
[... cut ...]
#1  pcl::transformPointCloud<sad::FullPointType, float> (cloud_in=..., cloud_out=..., transform=..., copy_all_fields=copy_all_fields@entry=true)
    at /usr/local/include/pcl-1.12/pcl/common/impl/transforms.hpp:232
#2  0x00007ffff79b6221 in pcl::transformPointCloud<sad::FullPointType> (copy_all_fields=true, transform=..., cloud_out=..., cloud_in=...)
    at /usr/local/include/pcl-1.12/pcl/common/transforms.h:237
#3  sad::LioIEKF::Align (this=0x555555615360) at /home/user/slam/slam_in_autonomous_driving-master/src/ch8/lio-iekf/lio_iekf.cc:79

后面代码由于需要run_frontend 生成"./data/ch9/keyframes.txt" 暂时运行不了
感谢高博耐心解答！同时我再换台电脑重新装个系统试试

gaoxiang12, 2023-03-28:
ok，我去看下run_mapping的问题

zerolover, 2023-03-28:
@lijh1024 你试试在 find_package(PCL REQUIRED QUIET) 的上一行加一句，因为不太清楚你自己编译的pcl有啥问题

```
set(PCL_DIR "/usr/lib/x86_64-linux-gnu/cmake/pcl") # use default pcl 1.10 in 20.04
find_package(PCL REQUIRED QUIET)
```

closed 2023-04-27 by lijh1024

gaoxiang12, 2024-02-07:
@YZH-bot 路径没给对？

tctctcc, 2024-04-16:
我在第四章也出现了类似的问题，./bin/test_preintegration 和 ./bin/run_gins_pre_integ 运行都会出现段错误，。。。
我看问题好像是根g2o或者Sophus有关