gaoxiang12/slam_in_autonomous_driving #60: Review of ch4

zerolover, 2023-04-05:
- 改成位移、速度、旋转更合适一些
[image]
- 公式错误
[image]
- 下标错误
[image]
- 这里的”加上“和1中的”加上“不是一个意思，但是同时出现在上下文中感觉不太合适
[image]
- 加粗
[image]
- 不同文献当中对残差的具体计算并不完全一致，甚至有的论文和实现**当中**也并不一 致。[去掉当中]
- 下标
[image]

gaoxiang12, 2023-04-06:
文字部分已经修正。
4.8和4.10这两个，因为bg和ba在i到j之间是认为不变的，所以下标用哪个都是一样的。。噪声项同理。。所以先不动了。

zerolover, 2023-04-06:
4.8 是括号的问题，没有下标的问题
4.10 的话，我理解 i,j 之间 ba,bg 是不变的(预积分是这么假设的，书上也提到了“我们固定 i 时刻的零偏估计”)，但是噪声项下标应该是不同的。其实就是把 4.10 标出来的三个 i 改成 i+1。
当然写成一样的也行，那就得把诸如 4.11 的噪声项 $\eta_{bg,k}$ 写成 $\eta_{bg,i}$

closed 2023-05-17 by zerolover