gaoxiang12/slam_in_autonomous_driving #63: ch3中关于欧拉角关于delta 角度的求倒

Vincent-ping, 2023-04-23:
求问一下，在ch3 eskf.hpp文件中，ObserveSE3函数里对偶拉角求倒不应该是Jl(-theta)吗？为什么会是单位阵，暂时没有高博的新书可以参考，有人能解释下公式3.66是怎么推导的吗

gaoxiang12, 2023-04-23:
这实际上是一个小trick：

[image]
[image]

Vincent-ping, 2023-04-23:
了解了，感谢高博，另外还有一个问题哈，同样在eskf.hpp中的ObserveWheelSpeed函数中，速度的雅可比矩阵除了对delta v是单位帧外，对姿态是不是是R× v的斜对称矩阵，因为这个vel_world不是通过姿态转到全局的吗？

gaoxiang12, 2023-04-23:
[image]

Vincent-ping, 2023-04-23:
大洋已经准备好了，赶快出版赶快买！！ ！  不过这句话轮速计并没有对R进行物理意义上的测量，这句话怎么理解哈？

gaoxiang12, 2023-04-23:
因为轮速这个传感器本身没有测量车辆的姿态这种用途啊。。它多数时候只有一个轮子的转速或者转动距离

closed 2023-04-23 by Vincent-ping