gaoxiang12/slam_in_autonomous_driving #67: When the code enables UI, the system freezes.

UAVlover, 2023-04-27:
Recently, I got a new computer installed with the latest version of Ubuntu 20.04.6. When running a program that includes "tools/ui/pangolin_window.h" to draw the UI interface, the system freezes. If the UI is disabled in the program, the program can run normally and print output in the terminal. Programs that directly call the Pangolin library for visualization can visualize and run normally, and it is currently unclear what the reason is.

gaoxiang12, 2023-04-28:
@UAVlover Yeah, same for my Yoga Pro 14s. I wonder if should be some hardware problems with OpenGL in the recent GPU cards.

UAVlover, 2023-04-28:
@gaoxiang12  Thank you for your response, Dr. Gao.  My new PC is a ThinkBook 16+ 2022, and I'm hoping that other guys who have same problem can write down their PC model here, which may help.

UAVlover, 2023-04-29:
@gaoxiang12  Hello, Dr.Gao. I have found a useful way to deal with this problem. 

1. sudo vim /etc/default/grub
2. find GRUB_CMDLINE_LINUX_DEFAULT="quiet splash"  
3. Then add "nomodeset" at the end, save, and exit. for example, GRUB_CMDLINE_LINUX_DEFAULT="quiet splash nomodeset" 
4. sudo update-grub
5. reboot
 After these operations, all programs can be executed well on my PC.

ForLoveandPeace, 2023-05-09:
Hi, you can `export LIBGL_ALWAYS_SOFTWARE=1`, instead of adding  "nomodeset" in /etc/default/grub. Hope it will help you.

UAVlover, 2023-05-09:
@ForLoveandPeace Thanks. In fact, I have tested this method as you mentioned. But, it didn't work on my PC.

DrZhongCH, 2023-05-15:
I think I installed NVIDIA GPU for DL previously.  Adding "nomodeset"  does not work for me. export LIBGL_ALWAYS_SOFTWARE=1   solve the problem.

UAVlover, 2023-05-15:
> I think I installed NVIDIA GPU for DL previously. Adding "nomodeset" does not work for me. export LIBGL_ALWAYS_SOFTWARE=1 solve the problem.

Thanks for your comment. It's wired , I installed nvidia driver too. Anyway, your problem has been solved ! Our discussion would be helpful for other guys!

closed 2023-05-28 by UAVlover

fongpd, 2023-11-18:
My PC is colorful yx p16,when I run the first program of the book，my system was freezed.

wzh506, 2024-04-04:
I tried expoert LIBGL_ALWAYS_SOFTWARE =1 but failed,I wonder how....

zjugaoming, 2024-07-08:
I was running the program in a virtual machine, and neither of the above solutions worked