gaoxiang12/slam_in_autonomous_driving #89: thanks for the nice opensource!

himhan34, 2023-06-12:
I am curious,are you going to make the book with English as well?
i always thankful for your commitment in this field. thanks!

gaoxiang12, 2023-06-12:
Yeah, Springer is already asking me to do the translation. But it takes time, and I'm still reviewing the Chinese version now. Maybe I'll start to do the translation in Sep or Oct of 2023.

himhan34, 2023-06-12:
Thanks for the kind review.
Then are you going to opensource the pdf as well? like before?

gaoxiang12, 2023-06-13:
There will probably be open-source and hard-copy versions from the publishing house like before because the hard-copy is really expensive for most of the readers.

himhan34, 2023-06-13:
thanks! i will wait till then. have a nice day!

closed 2023-06-13 by himhan34