gaoxiang12/slam_in_autonomous_driving #92: Docker compilation

Alwen-Yan-edu, 2023-07-03:
Thanks to Dr. Gao for his contribution. And I have something confusing: When I compiled the project in docker mode according to readme, I found that I still need to compile g20 and Pangolin, but I have compiled these two libraries in Dockerfile. Why?

gaoxiang12, 2023-07-03:
@Alwen-Yan-edu The Docker support is merged from PR #91, and I have not tested it yet. The project looks for g2o and pangolin in the thirdparty/ directory. So if you want to set it to any other places, you should modify the corresponding CMakeLists.txt.

Alwen-Yan-edu, 2023-07-03:
I have finished compiling the container and got the executable file. But I got an error at runtime. :<
[image]

closed 2023-07-26 by Alwen-Yan-edu

xhKuo, 2024-03-21:
> I have finished compiling the container and got the executable file. But I got an error at runtime. :< ![FJU)WQGXR`35 (53$_I8) Q](https://private-user-images.githubusercontent.com/57043989/250521059-921f76b9-0f4f-4242-9904-f9c425932dd4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEwMTU5NzQsIm5iZiI6MTcxMTAxNTY3NCwicGF0aCI6Ii81NzA0Mzk4OS8yNTA1MjEwNTktOTIxZjc2YjktMGY0Zi00MjQyLTk5MDQtZjljNDI1OTMyZGQ0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIxVDEwMDc1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBlMTIxMTBmYjk1NmVmZjI1Y2UxN2QyOGI2MDVlOTQyNDVjMjE1Y2JjZDQ2NDgxM2Q0YmE5ODRjN2EwMzBkZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.VLn1xidv0gCgWkaPQ42xdwGLUpZog4ysMPS1DEMhfGs)

Hello, I also encountered the same problem. Could you please tell me how you finally solved this problem? I have installed G2O and the pangolin library in /usr/local
[image]

kktfox, 2024-04-03:
> I have finished compiling the container and got the executable file. But I got an error at runtime. :< ![FJU)WQGXR`35 (53$_I8) [...]

Did you solve the problem?

kktfox, 2024-04-03:
use this: 
`
export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH
`
can solve the question