gaoxiang12/slam_in_autonomous_driving #97: iekf 公式8.7a det_xk计算的疑问

xubin1991, 2023-07-23:
书中的公式是det_xk=K_k(z-h(x_k))相当于是迭代很多次，我理解等价于观测重复使用了很多次。

我在知乎上找到iekf另一个版本：
https://zhuanlan.zhihu.com/p/141018958

[image]

在算r的时候相当于限定在x0处，只是迭代了H矩阵，观测等价于只使用了一次。

这两种表达似乎是冲突的，哪一个是正确的（或者更有道理呢？）

gaoxiang12, 2023-07-23:
每次迭代时最近邻也不一样，观测维度并不一样，并不能理解成观测重复使用多次。应该理解成在不同工作点上多次建立了不同的观测模型。
这是因为LIO里的观测方程实际上是ICP或者NDT导致的，和通用的IEKF并不一样。其他材料里的IEKF并不需要重新找一遍点云最近邻。

xubin1991, 2023-07-23:
感谢高老师的解答。

我感觉这个地方还是值得商榷的，假设一种极端状况每次最近邻找到的点都一样，那么按照书中公式，无论起始x在何处，只要迭代次数够多，x_k就会收敛到完全吻合。

最终看来虽然P的更新是对的，但是x已经移动的很大了，而且这个移动步长只和迭代次数有关，与x的初始P弱相关（即无论多大的P只要移动步长够长就能移动很大的x）。

感觉这里有点不合理，而如果使用原始iekf的公式则不会有问题，因为每次更新x的时候会考虑当前x_k离初始x_0的移动距离，因而最终的移动距离是和初始P是强相关的。

gaoxiang12, 2023-07-24:
点云配准的意义就是利用多次迭代来让算法收敛……如果一次就收敛了，IEKF根本不会继续迭代，或者反过来说，如果x移动较远，也不可能让最近邻完全不变。
而且我不太理解你说的P弱相关的问题，因为IEKF的dx计算也会有P的参与（实际是参与了K的计算）。初始的P会投影到当前工作点上并影响dx的计算。只是跳过了每次迭代时的P阵往后递推的过程。

xiaohai-AI, 2023-09-06:
I think for traditional IEKF, the iteration only happen to the state translation equation, for measuring equation NO iteration, its a instant mapping process from state to measured variable. Also the linear Taylor series expansion is already a approximation, AND repeat use the z value for 0 - (i-1) is the second approximation
but here in the book, you can think our observing equation is also a iteration equation, but this iteration happen in another dimension,but which is correlated with the state equation iteration dimension

qwer23579, 2023-09-18:
对于公式8.7-8.8中，协方差P(k)到P(k+1)的投影计算没有疑问。这个一直在改变工作点，更精确线性化。
但是对于x(k)到x(k+1)的计算是值得讨论的。EKF的增益k就是对于预测与测量的融合比例，公式8.7迭代过程一直在基于x(k)与测量z进行比例融合，预测退化成了一个初始惩罚项。虽然对于最终结果影响不大或者匹配收敛更稳定，但不符合EKF对K的含义。
@xubin1991 反馈的传统IEKF，迭代改变的是测量的线性工作点。但是对于x(k+1)的计算，是由预测x0与当前线性化的测量融合得到的。
@gaoxiang12

houchangmeng, 2023-10-09:
我想问一下这个H阵到底是对误差状态deltax的雅克比，还是对x的雅克比，看书上应该是对x的雅克比，我也能理解，但是网上有一些资料为什么说是对误差状态的雅克比。