gaoxiang12/slam_in_autonomous_driving #122: 如下是运行您的代码的时间：

jackzyh, 2023-10-20:
高博您好：如下是为测试第五章遇到的问题
jack@jack-HP-ZBook-15-G3:~/Desktop/slam_in_autonomous_driving-master/bin$ ./test_nn 
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from CH5_TEST
[ RUN      ] CH5_TEST.KDTREE_KNN
Failed to find match for field 'intensity'.
Failed to find match for field 'intensity'.
I1020 17:57:48.238508 14694 sys_utils.h:32] 方法 Kd Tree build 平均调用时间/次数: 11.5117/1 毫秒.
I1020 17:57:48.238765 14694 test_nn.cc:228] Kd tree leaves: 18869, points: 18869

[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from CH5_TEST
[ RUN      ] CH5_TEST.KDTREE_KNN
Failed to find match for field 'intensity'.
Failed to find match for field 'intensity'.
I1020 17:54:21.204362 13088 sys_utils.h:32] 方法 Kd Tree build 平均调用时间/次数: 594.171/1 毫秒.
I1020 17:54:21.204608 13088 test_nn.cc:228] Kd tree leaves: 18869, points: 18869

运行效率天差地别，效率高的是代码里面把livox-ros-driver包编译进去了，但是cmake里并没有include编译出来的两个头文件，非常玄学，证明与livox-ros-driver包其实没有本质联系。但是一定要把 livox ros driver包 add_subdirectory(thirdparty/livox_ros_driver) 编译进去算法效率才会没有问题。

但是如果在ROS2环境下，把ROS2版本的 livox ros driver包编译进去算法效率还是非常慢。

如果clone高博开源整套代码里面，在代码中把livox相关处理函数注释掉，livox-ros-driver包不编译进去，就会发现代码每一个模块都运行效率特别慢，没有体现并发的优势，但是编译不会报错。

gaoxiang12, 2023-10-20:
计算机应该不会有玄学。。我觉得你可能是编译了一个debug模式。
你可以把build删除，然后用cmake .. -DCMAKE_BUILD_TYPE=Release重新编译一遍再试试。

jackzyh, 2023-10-21:
感谢高博 问题已经解决 谢谢！

> 计算机应该不会有玄学。。我觉得你可能是编译了一个debug模式。 你可以把build删除，然后用cmake .. -DCMAKE_BUILD_TYPE=Release重新编译一遍再试试。

感谢高博 问题得到解决 谢谢!

closed 2023-11-11 by jackzyh

Chaoyue-cpu, 2024-03-28:
你好，你运行第五章的代码还有那个找不到intensity吗。我删除build还是有这个问题

jackzyh, 2024-03-28:
这是来自QQ邮箱的假期自动回复邮件。你好，我最近正在休假中，无法亲自回复你的邮件。我将在假期结束后，尽快给你回复。