gaoxiang12/slam_in_autonomous_driving #123: ch2编译报错

sja3015, 2023-10-30:
有大佬 看看我这个小问题吗
[image]

bear-rabbit, 2023-11-03:
我也遇到了！我看csdn上的一个博客解决啦https://blog.csdn.net/ht_lf?spm=1010.2135.3001.5343

iFly0523, 2024-02-05:
为啥我在slam_in_autonomous_driving-master目录总体编译了，还是会报和上面一样的error呀。我的是18.04的版本？恳求大佬指导

sja3015, 2024-02-05:
我的是20.04 我之前用18.04总会报奇奇怪怪的问题  改成20.04就没出现过了

enguang2, 2024-03-01:
Don't run the Cmake and make command under ch2 directory (actually not under any chapter directory), **Please run the Cmake command under the main directory**. Look at the CMakeList.txt, which already adds src as a sub directort and then Cmake would do the compilation in an iterative way. The common/eigens_type.h missing is due to the compilation missing of commons folder.