gaoxiang12/slam_in_autonomous_driving #132: 第三章 ESKF 噪声Q的疑问

Maxinjun, 2023-12-14:
代码如下： 
       double ev = options.acce_var_;
        double et = options.gyro_var_;
        double eg = options.bias_gyro_var_;
        double ea = options.bias_acce_var_;

        double ev2 = ev;  // * ev;
        double et2 = et;  // * et;
        double eg2 = eg;  // * eg;
        double ea2 = ea;  // * ea;

        // 设置过程噪声
        Q_.diagonal() << 0, 0, 0, ev2, ev2, ev2, et2, et2, et2, eg2, eg2, eg2, ea2, ea2, ea2, 0, 0, 0;

第一个疑问是：应该把上面的注释放开，需要平方
第二个疑问是， ev2、 et2需要乘以dt^2,  eg2、ea2需要乘以dt， 参考公式如下
[image]
根据代码
              // 噪声由初始化器估计
              options.gyro_var_ = sqrt(imu_init.GetCovGyro()[0]);
              options.acce_var_ = sqrt(imu_init.GetCovAcce()[0]);
可以得知acce_var_是静止时离散时间的加速度计噪声标准差，单位是m/s^2;   如果ev2=et=options.acce_var_， 即Q中速度对应的方差的单位也是m/s^2, 这明显是不对的，如果是 acc_var^2*dt^2, 单位为 m^2/s^2, 这才是正确的

925779599, 2025-06-27:
我也同样的疑问，有没有大佬解释下。

owlright, 2025-12-18:
我用自己录的bag包测试，解开注释的效果要比原本的好一些，虽然也不是很准。

下面第一张图是原本的程序，第二张是解开注释的。

[image]

[image]