gaoxiang12/slam_in_autonomous_driving #133: 有人使用Qt跑rosbag遇到我这样的问题了吗？

teachermaDSM, 2023-12-14:
使用qt直接运行的时候由于没有source /opt/ros/melodic/setup.bash，运行带有rosbag的包会报错；
`
[rospack] Error: package 'rosbag_storage' not found
[librospack]: error while executing command
terminate called after throwing an instance of 'pluginlib::ClassLoaderException'
  what():  Unable to find package: rosbag_storage
`
请问这个该如何解决？
ps:在自己的终端中直接./bin/test_lidar_io没有问题

teachermaDSM, 2023-12-14:
需要使用Qt进行Debug，因此请教大家如何配置QtCreator才可以解决该问题。。。

qiaoyaya2011, 2024-10-10:
你好，想问你怎么解决的，我目前用vscode进行debug也遇到这样的问题，最后怎么解决的啊。

qiaoyaya2011, 2024-10-11:
[rospack] Error: package 'rosbag_storage' not found
[librospack]: error while executing command
terminate called after throwing an instance of 'pluginlib::ClassLoaderException'
  what():  Unable to find package: rosbag_storage

teachermaDSM, 2024-10-15:
> 你好，想问你怎么解决的，我目前用vscode进行debug也遇到这样的问题，请问最后怎么解决的啊。 @teachermaDSM

没有特别好的办法，原因我想应该就是没有source那个ros对应的setup.bash导致rosbag无法找到，我后面都是用打印log来debug的，希望高翔老师能给些debug含有rosbag的方法