gaoxiang12/slam_in_autonomous_driving #134: /usr/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:1320: undefined reference to `pcl::visualization::PCLVisualizer::createActorFromVTKDataSet(vtkSmartPointer<vtkDataSet> const&, vtkSmartPointer<vtkLODActor>&, bool)

Qidayourfather, 2023-12-27:
/usr/bin/ld: CMakeFiles/point_cloud_load_and_vis.dir/point_cloud_load_and_vis.cc.o: in function `bool pcl::visualization::PCLVisualizer::fromHandlersToScreen<pcl::PointXYZI>(pcl::visualization::PointCloudGeometryHandler<pcl::PointXYZI> const&, pcl::visualization::PointCloudColorHandler<pcl::PointXYZI> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, Eigen::Matrix<float, 4, 1, 0, 4, 1> const&, Eigen::Quaternion<float, 0> const&)':
/usr/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:1320: undefined reference to `pcl::visualization::PCLVisualizer::createActorFromVTKDataSet(vtkSmartPointer<vtkDataSet> const&, vtkSmartPointer<vtkLODActor>&, bool)'
/usr/bin/ld: CMakeFiles/point_cloud_load_and_vis.dir/point_cloud_load_and_vis.cc.o: in function `bool pcl::visualization::PCLVisualizer::addPointCloud<pcl::PointXYZI>(pcl::PointCloud<pcl::PointXYZI>::ConstPtr const&, pcl::visualization::PointCloudColorHandler<pcl::PointXYZI> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)':
/usr/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:146: undefined reference to `pcl::visualization::PointCloudGeometryHandlerXYZ<pcl::PointXYZI>::PointCloudGeometryHandlerXYZ(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZI> const> const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [src/ch5/CMakeFiles/point_cloud_load_and_vis.dir/build.make:206：/home/yuzq/learn_resource/learn-slam-in-autonomous-driving/slam_in_autonomous_driving/bin/point_cloud_load_and_vis] 错误 1
make[1]: *** [CMakeFiles/Makefile2:4208：src/ch5/CMakeFiles/point_cloud_load_and_vis.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....

有朋友遇到这个问题了吗？救一下！！

MOLUPENGYOU, 2024-04-12:
请问一下解决了吗？我也碰到了一样的问题。

MOLUPENGYOU, 2024-04-14:
我之后找到的解决方式是：cp PCL /build/lib/* to /usr/local/lib

aizhiluo, 2024-07-17:
> 我之后找到的解决方式是：cp PCL /build/lib/* to /usr/local/lib

能具体解释下怎么解决的吗