gaoxiang12/slam_in_autonomous_driving #138: IEKF 卡尔曼增益后协方差矩阵的投影问题

1344618323, 2024-01-18:
https://github.com/gaoxiang12/slam_in_autonomous_driving/blob/fe9c79bf03cffb4b9e6388015a42c747c318a182/src/ch8/lio-iekf/iekf.hpp#L254

在update P那一步，`cov_ = (Mat18T::Identity() - Qk * HTVH) * Pk` 用得是上一步迭代（最后一个线性化点）后的协方差矩阵`Pk`，那么project P中计算 dtheta时应该用最后一次 `dx_` 里的 $\theta$ 而不该是 `(R_.inverse() * start_R).log()`？

ZYXiao-g, 2024-07-03:
同样发现这个问题了， 第三章eskf代码就是采用dx_, 到第8章代码就是用的R_.inverse() * start_R.从公式看确实是第8章代码有问题，在迭代的时候也是采用的R_.inverse() * start_R也是有问题的

ZYXiao-g, 2024-07-03:
并且最后一步应该也是有问题的，更新完对P重新投影的时候，应该是切空间k->k+1的投影，而代码里面是0->k+1时刻的投影。所以迭代收敛完成的时候fast_lio里面直接是忽略掉这一步也是有道理的，此时的值非常小。

gaoxiang12, 2024-07-03:
这算是一个trick。
本来每一次IEKF迭代的时候都要做一次切空间投影，这样能够实时得到投影后的P，也就是k->k+1的投影。
但实际上并没有这个必要，因为每次迭代重点是更新均值x并重新计算最近邻，而P并不参与这部分计算。所以可以在迭代结束时再计算P的投影，于是就像现在第8章实现那样，算完整个迭代再进行投影了。

ZYXiao-g, 2024-07-04:
[image]

这里的疑问不是迭代过程中疑问，是最后迭代结束后切空间投影存在疑问，具体在图片中展示

dongbo19, 2024-09-10:
I believe it should be the transformation from the 0th iteration to the kth iteration, rather than the transformation from the kth to the (k+1)th iteration. You can refer to the paper '**_Notes on Kalman Filter (KF, EKF, ESKF, IEKF, IESKF)_**', which provides a detailed introduction to the theory of IESKF.

zubingtan, 2024-11-18:
> ![选区_658](https://private-user-images.githubusercontent.com/52555001/345690645-f385f69e-9d7b-4cd3-b2c3-4008aa18fab8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE5MDk4MTksIm5iZiI6MTczMTkwOTUxOSwicGF0aCI6Ii81MjU1NTAwMS8zNDU2OTA2NDUtZjM4NWY2OWUtOWQ3Yi00Y2QzLWIyYzMtNDAwOGFhMThmYWI4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTE4VDA1NTgzOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU0ZGE4ZmNlNjZlZGJkNDE0YzJlNTE1YzU1MDZmNDBmMDlmZGJjYmZkZDQ4YjI4YzQ5OGM4OTlhYzlmOGMyN2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.l5sHQymWo6Mgl-gjJZu4j22D2WdLpTn1s6Y2r7PCyVI)
> 
> 这里的疑问不是迭代过程中疑问，是最后迭代结束后切空间投影存在疑问，具体在图片中展示

书上的是 dx，也就是k到k+1的投影；

@dongbo19 您好，想问一下  'Notes on Kalman Filter (KF, EKF, ESKF, IEKF, IESKF)' 书中何处指明 “it should be the transformation from the 0th iteration to the kth iteration” 呢？希望指明一下方向。我看到的是最后甚至都没有整体（cc @gaoxiang12 ）
[image]