gaoxiang12/slam_in_autonomous_driving #139: Ch3,ESKF 中 IMU的坐标系问题

zvxvd, 2024-02-01:
请问在ESKF 融合时，IMU 应该采用哪种坐标系呢，我现在的IMU 为 前（X+），左（Y+），上（Z+），在使用我自己的数据进行融合时，发现估计的速度vel_x 和 vel_y 与实际速度差别很大，但是轨迹感觉是对的

下面是可视化的轨迹：绿色为RTK测量得到的原始位置，红色为IMU预测得到的位置
[image]

下面是pangolin 中 vel_x vel_y vel_z 和 baselink_vel_x ,baselink_vel_y ,baselink_vel_, 差别非常大

[image]

[image]

[image]

gaoxiang12, 2024-02-04:
pangolin一个是本体下速度，一个是世界系下速度，中间实际相差一个旋转矩阵。当然这么说不严格，应该说是“世界速度矢量在本体系下的坐标表示”。

我看上去你的结果应该没问题的。

zvxvd, 2024-02-04:
好的，谢谢高博

closed 2024-02-04 by zvxvd