gaoxiang12/slam_in_autonomous_driving #149: Request for Korean translation

changh95, 2024-03-29:
Dear Dr. Gao,

I hope this message finds you well. I'm Hyunggi Chang, a SLAM engineer at StradVision in South Korea and leader of the Spatial AI KR community, which includes around 4000 SLAM enthusiasts and experts.

I want to express my gratitude for your book, which has become a staple for SLAM research in Korea. I was part of a team that translated the slambook into Korean unofficially in 2018, and it was a rewarding experience that helped many here, including myself.

I saw that the slambook has been officially published in Korean, which is great news for our community. Your work is highly valued here, and I thank you again for your contributions.

I would like to offer my help with the Korean publication of your next book on slam-in-autonomous-driving. My experience in visual and lidar SLAM and familiarity with the terminology could be beneficial.

If there's a better way to discuss this, please let me know. Thank you for your time and consideration.

gaoxiang12, 2024-03-30:
Hello @changh95 ,
It's great to hear that you support my book. I wanted to let you know that I'm currently working on an English version of the SAD book, which you can find at this URL: https://github.com/gaoxiang12/slam-in-ad-en. If you're interested in receiving a copy of the Chinese version, please provide me with an email address so that I can send it to you.

changh95, 2024-03-31:
Dear Dr. Gao,

Thank you very much for your prompt response! I'm delighted to learn about the English version of the SAD book you're currently working on. I believe cross-referencing this with the Korean translation will be immensely beneficial, ensuring both versions are as accurate and valuable as possible. During this cross-checking process, I'll also be on the lookout for any potential improvements in the English translation and will be more than happy to share my findings with you.

Please find my email address below to facilitate the sharing of the Chinese version and any future communications: hyunggi.chang95@gmail.com

I'm looking forward to contributing to this exciting project and enhancing the accessibility and quality of your work for a wider audience.

Thank you once again for your time and consideration.

changh95, 2024-04-06:
> Hello @changh95 ,
>
> It's great to hear that you support my book. I wanted to let you know that I'm currently working on an English version of the SAD book, which you can find at this URL: [...]

@gaoxiang12 

Sending you a notification, in case you missed my last message  😄 

Thank you again!