gaoxiang12/slam_in_autonomous_driving #161: 公式4.7的疑问

tzccg0, 2024-05-29:
4.7c公式中间是不是还应该减去一个g * △tij * `△tij，即

[image]
由4.7b，可以得出
[image]

4.6c、4.7d和4.7c的公式是不是差了一项g * △tij * `△tij
这儿不是特别理解

tzccg0, 2024-05-30:
如果4.6c的公式改成这样子，应该就能对上了
[image]

h-k8888, 2024-06-03:
@tzccg0 是4.6c的公式写错了，我自己也看了很久。
你可以从4.4c看，对g的积分就是整个时间间隔，而不是一个个delta t的平方相加。
其实4.7c很简单，就是把v_k变成v_i + v_ik。
希望在新版的书里，翔哥能够改正这一点。
@gaoxiang12

gaoxiang12, 2024-06-04:
[image]

目前版本已经修改了，各位可能拿到的是最早版本的书。

h-k8888, 2024-06-04:
> <img alt="image" width="781" src="https://private-user-images.githubusercontent.com/6635511/336264402-efa8166d-c73e-498a-a29d-d8b003ca3b3d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc0Nzg1ODIsIm5iZiI6MTcxNzQ3ODI4MiwicGF0aCI6Ii82NjM1NTExLzMzNjI2NDQwMi1lZmE4MTY2ZC1jNzNlLTQ5OGEtYTI5ZC1kOGIwMDNjYTNiM2QucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDRUMDUxODAyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Yzc1YWQ0MzE5ZDExZTljYmM4NzBkZGJlZjIwZDA3YTRhNjdmM2NhM2IxODZkZTRjMzU2NjRiNWIxNmVjNTVhNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.mEHEj9vCTn6ZFtPWbMudi0RDqaxrIoilKuZrBzF2txY">
> 目前版本已经修改了，各位可能拿到的是最早版本的书。

4.6c中的第三项，对重力g在i至j时间间隔的积分，写成这个样子应该有歧义吧。
delta_t^2的求和，应该不能等于t_ij的平方吧。
从公式推导的角度来说，没办法变成4.7c中对g积分的形式。

tzccg0, 2024-06-05:
@gaoxiang12 我这边看的应该就是最新版，和您贴出来的截图是一样的
我这边疑惑的点是，根据4.7b等式，红色部分vi和delta_v_ik的计算，应该需要要减去一个g * delta_t_ij才能成立
高老师有空看到的话，能不能帮我解个惑，是不是我对公式的理解还不到位
[image]

tzccg0, 2024-06-06:
回去仔细想了下，参考了 @h-k8888 的疑问，大概弄清楚了
[image]
还是学艺不精了，看书也还不够仔细，需要继续努力啊