gaoxiang12/slam_in_autonomous_driving #195: ch2 EKF 公式2.116  疑惑

gyyc233, 2025-04-09:
EKF 更新方程 P_k的计算，为啥使用C_k 而不是H_k

[image]

RicoJia, 2025-04-11:
这个在最开始的线性KF里边就是用的C_k. 我有一个[blog post](https://ricojia.github.io/2024/03/15/robotics-foundamentals-kalman-filter/) 证明Kalman filter 可以最小化状态估计的covariance, 同时他对`x_k_hat` 的状态估计是最大化`p(x_k | x_{k-1})` 和 `p(z_k | x_k)`的共同分布。数学有点复杂，可以看看知乎

gyyc233, 2025-04-11:
> 这个在最开始的线性KF里边就是用的C_k. 我有一个[blog post](https://ricojia.github.io/2024/03/15/robotics-foundamentals-kalman-filter/) 证明Kalman filter 可以最小化状态估计的covariance, 同时他对`x_k_hat` 的状态估计是最大化`p(x_k | x_{k-1})` 和 `p(z_k | x_k)`的共同分布。数学有点复杂，可以看看知乎

谢谢你的分享，但我看的多数EKF(扩展卡尔曼)文章（包括你的blog）里面协方差 P_k的更新大多用的都是H_k（$\frac{\partial h}{\partial x}$）。

我提的问题源自纸质书上的公式编号，或许它在后续版本的书上已经更新了，但是我不确定，因为在之前的issues里面没有找到相同内容

closed 2025-04-23 by gyyc233